ThunderFly-aerospace/PX4-FlightGear-Bridge #4: Multi-vehicle simulation

slimonslimon, 2020-04-12:
For multi vehicle simulation we need solve folowing problems:
 
- [ ] multiple FG can connect to same FG server, but every FG need pick port for incoming comunication (fg param) 
- [ ] bridge and FG need this comunication to pick ports somehow (fg param, and bridge inside)
- [ ] bridge to PX4 need setup port for Communication - (vehicle can setup listening port in its boot, bridge inside)

Jaeyoung-Lim, 2020-04-12:
@slimonslimon For the port configuration, [this](https://github.com/PX4/Firmware/blob/master/Tools/gazebo_sitl_multiple_run.sh) script might be useful to compare how it is done for gazebo

slimonslimon, 2020-04-13:
Thank you for your comment!
Our solution for FG will work in little bit other way, because we need spawn multiple types of vehicles.

I'm close to working example now. 

@Jaeyoung-Lim  Do you have some experiences with communication and orchestration of multiple vehicles? And are you able to run current Px4<->FG sitl?

slimonslimon, 2020-04-13:
Now, there is an experimental branch flightgear-multi, which supports: 
`make px4_sitl_nolockstep flightgear_plane___multi`
And you are able to run it mutiple times. You only need connect FG to server manualy, for now. (It will be automated later.)

There is one other problem with selecting "empty" "px4id" - now it counts running px4 instances - this can work only on first run. (You cannot restart any instance separetly)

Jaeyoung-Lim, 2020-04-13:
@slimonslimon I would prefer moving the multivehicle sim to a separate script, rather than making it part of the make target (e.g. the script link I linked above) so that we can be more versatile on the number of vehicles.

For example, for gazebo you can specify arbitrary number of vehicles and targets through the script

slimonslimon, 2020-04-13:
Thank you for your comment! I will improve this later.

slimonslimon, 2020-04-29:
last problem of selecting PX4 ID was passed to user start script - it should be set by enviromental variable in experimental branch 

but this problem is not part of FG bridge, It is problem of start scripts. 

Brdige has multi-vehicle support.

closed 2020-04-29 by slimonslimon