ThunderFly-aerospace/PX4-FlightGear-Bridge #5: What is the correct way to install the models?

Jaeyoung-Lim, 2020-04-13:
Is there a documentation on how to correctly install the models of flight gear?

I was trying to get started (Haven't looked at any additional documentation) and am stuck here. 
I guess everyone else will wonder the same thing when they get started

Thanks,

slimonslimon, 2020-04-13:
Okey.

You install FlightGear probably from repositary (most straght way). But you dont install fg-data package. When you install it (or download it separetly) FG needs know, where this data folder is.
(probalby: /usr/share/games/flightgear ) - use "choose folder" in this dialog.

Test your FG instalation by running command "fgfs" - it will show this dialog, or start game with plane Cesna.

After that you can download vehicles (link in readme) and extract douwloaded zip file into Aircraft folder (probably:  /usr/share/games/flightgear/Aircraft), exactly subfolder in FG-Data package

Can you tell me, what distribution and version of Linux are you using?

Jaeyoung-Lim, 2020-04-13:
@slimonslimon I am using Ubuntu 18.04, and I installed flight gear through 

```
apt install flightgear
```
There is no package called `fg-data` in apt. Where can I get `fg-data`?

I can certainly run fgfs. Would it be possible to maybe automate the model addition? Not sure how big of a task this is.

slimonslimon, 2020-04-13:
Automated model downloading is pending issue (i did not setup issue on github yet)

But it seems that in version of FG 2018.1.1 work something in other way that I expected. (It is not complining abou tmodel but about whole data folder) -  I will try reproduce your error.

One other "stupid" question: do you have last version of "flightgear" branch? (I made today an update of scripts)

slimonslimon, 2020-04-13:
we have tested this only with FG 2019.1.1 (and 2019.2.0 - with some updates)

slimonslimon, 2020-04-13:
@kaklik can you try reproduce this error (FG 2018.1.1 from ubuntu 18.04 repo?)

roman-dvorak, 2020-04-13:
Yes, I can confirm this problem. 

You must install flightgear from flightgear PPA. It will also install FGdata 
```
sudo add-apt-repository ppa:saiarcot895/flightgear
sudo apt update
sudo apt install flightgear
```
_(more_ informations: http://wiki.flightgear.org/Howto:Install_Flightgear_from_a_PPA)_

You'll probably have to do it:
```
sudo apt install flightgear --upgrade
```

kaklik, 2020-04-13:
> @kaklik can you try to reproduce this error (FG 2018.1.1 from ubuntu 18.04 repo?)

I do not have any setup with this old FG version.  I think we should not support it.  I suggest to support at least 2019.1.1 FG version and later.

slimonslimon, 2020-04-13:
but it works correctly at FG 2016.x.x in Debian 9 
I will investigate this problem tomorrow.

Jaeyoung-Lim, 2020-04-14:
@slimonslimon @kaklik @roman-dvorak Thanks!

I will try to update flightgear and try again.

Jaeyoung-Lim, 2020-04-25:
@kaklik @slimonslimon @roman-dvorak I have tried the rascal vehicle with flightgear 2019.1.1 on Ubuntu bionic. 

However, it results in a flyaway in misison mode. 
[image]

kaklik, 2020-04-26:
The flyaway behavior seems to be a bug of PX4. It happens in the case where you do not have any mission waypoint selected. (I do not really know how this algorithm working)

The expected and correct behavior is the following: 

[image]

Installation of models we suppose to solve by the creation of a [FlightGear hangar 
](http://wiki.flightgear.org/Hangar_catalog) with vehicles adapted to UAV SITL development.  There seem to be slightly different requirements than standard simulated aircraft. 

The Protocol directory permissions have been resolved by recent commits to FlightGear bridge and there should be no requirement to manually change permissions. 

Yes, the documentation is mainly uncomplete. We [have started to make the documentation](https://github.com/ThunderFly-aerospace/PX4Devguide/blob/b568f79e9541c871b6d3b7f570a609c98e03f629/en/simulation/flightgear.md). But it will take some time to get it complete.  But testing is welcomed during this period. :)

@Jaeyoung-Lim is it possible to link this issue or [the pull-request](https://github.com/PX4/Firmware/pull/14539) from the description of your video to achieve consistency in explanation of FlightGear bridge usage?

Jaeyoung-Lim, 2020-04-26:
I understand the long term goals of the nice architecture, but if I cannot run it "out of the box" following the documentation, it means not much people using px4 will struggle to run flight gear. This is a problem if we want to get the bridge merged as PX4. I don't expect the full documentation to be complete, but even the readme of this repo are missing instructions.

I believe this issue is basically the issues I am running into by "testing" the documentation of the readme of this repo, which I am struggling to run properly. 

@kaklik I am not sure what you mean by "mission waypoint selected" the same mission works well in SITL gazebo, and it is being checked in the CI. Does this mean you have seen this before? or does this mean it is not possible to test mission mode in flight gear? I cannot reproduce this issue in SITL gazebo

Jaeyoung-Lim, 2020-04-26:
I also noticed the following errors,

```
ERROR [sensors] Sensor Accel #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Accel #0 priority: 1
WARN  [sensors] Remaining sensors after failover event 0: Accel #1 priority: 0
WARN  [sensors] Remaining sensors after failover event 0: Accel #2 priority: 0
ERROR [sensors] Gyro #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Gyro #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Gyro #0 priority: 1
WARN  [sensors] Remaining sensors after failover event 0: Gyro #1 priority: 0
WARN  [sensors] Remaining sensors after failover event 0: Gyro #2 priority: 0

```

slimonslimon, 2020-04-26:
@Jaeyoung-Lim at first Thank you for your testing and suggestions.

I will test the documentation today - it will be improvet to be correct at ubuntu.

To your errors:  What is you FPS in FlightGear?  (you can select it from View->View Options-> Show frame rate)?

To the mission problem: It seems there is an problem when Take off point is at opposite side than direction of aeroplane at ground. But QGC shows correct position of plane, so I think, that PX4 knows, where the plane is, but does not turn it. I dont know why?

To Flightgear version etc. It probably should work with Distribution version, but it does not in ubuntu. I need to look deeper at this problem, but fist I need to install ubuntu on my machine.

kaklik, 2020-04-26:
@Jaeyoung-Lim  @slimonslimon  I think it will be better if you split the discussion to several separate issues for each topics.  I think this issue should be left for model installation as possible.  (I want to close is as soon as I make the hangar.xml for FlightGear.)

Jaeyoung-Lim, 2020-04-26:
@slimonslimon Thanks for clarifying, indeed moving the takeoff waypoint fixed the issue.
- Video: https://youtu.be/YjqC2qFbHbE

Interestingly, this is not reproducable in SITL gazebo. The waypoint tracks properly even though the takeoff waypoint is at the back of the vehicle when armed. 

@kaklik Thanks, makes sense. I will open separate issues since now I know that the whole workflow is working.

Jaeyoung-Lim, 2020-04-26:
@kaklik Seperated flyaway issue to https://github.com/ThunderFly-aerospace/PX4-FlightGear-Bridge/issues/10

slimonslimon, 2020-04-29:
This base of this issue was solved by additing folder with models (as git submodules). We found, that most models need to be somehow costumised for our use.

Now, the bridge will have models as subfolder and startup scripts will point FG to search vehicles in this folder.

closed 2020-04-29 by slimonslimon