ThunderFly-aerospace/PX4-FlightGear-Bridge #10: Fixed wing mission sometimes result in a flyaway

Jaeyoung-Lim, 2020-04-26:
I have experienced that sometimes fixed wing missions result in a flyaway

[image]

This seems to have somekind of relation ship with where the takeoff waypoint is located related to the vehicle, but this is not consistent. Sometimes the same mission results in a flyaway / or not.

Tested on Ubuntu Bionic, Flightgear 2019.1.1

slimonslimon, 2020-04-26:
hmmm... Im not able to reproduce this issue after last upstream/master merge.

Jaeyoung-Lim, 2020-04-26:
@slimonslimon What do you mean by upstream/master merge? 

I have tried the branch on https://github.com/PX4/Firmware/pull/14539, just now, but the problem still persists.

Here is the log: https://review.px4.io/plot_app?log=c26ab6d6-c36e-45d3-8178-d18044b2c748

slimonslimon, 2020-04-27:
I tested it about 10 times, but I'm not able to setup mission, which fails. I will try this at evening.

Please, look at PX4 console, is there any message like: 
```INFO  [commander] Takeoff detected```

I have much more logged messages in my log (but I'm not sure, it can by by some modifications, of our review, or by some log settings - Im no so familiar with this.):
http://review.thunderfly.cz/plot_app?log=74ba0769-2618-476d-a078-a0e24473d768

Jaeyoung-Lim, 2020-04-27:
@slimonslimon which environment are you testing in? Could it be something different coming from the system?

slimonslimon, 2020-04-27:
I forgot to say, that before I saw this situation on my PC..... and @kaklik has the same situation. So I think it should not be something from system. 

I have debian 9.x, gcc 6.x, FG 2019.1.1 or Next (aka 2019.2) both from source. (distribution version is old 2016.x.x, I can test it on this version too) 
Maybe, I use old version of QGC (3.5.1)

Generally, The PX4 is connected to FG over the bridge, and it is consuming so few information from its property tree, so it should work with most version of FG. The most problems comes from low FPS of FG which limits frequency of messages from FG.
But FG is full of "easter egg" without any direct documentation, so very offen something stop work and than magicly start to work. (I spent half hour today by searching why TerraSync stop work, and finally it starts work with same set of parameters as I have before)

Now more about this problem, how looks the mission plan? Depend the situation on it?

Jaeyoung-Lim, 2020-04-27:
@slimonslimon I can share you the mission file, but for me it happens all the time. So I think it maybe that it doesn't depend "only" on the mission.

The FPS for me ranges from 80 - 150. I looked at the flightgear bridge code and this should be sufficient. 

It seems like you and I are running a different flight gear distribution, since I didn't build it from source. I used the distributed version from 
```
sudo add-apt-repository ppa:saiarcot895/flightgear
sudo apt update
sudo apt install flightgear
```

Jaeyoung-Lim, 2020-04-27:
@slimonslimon Since this is not working with bionic, could you also document on which version combination this has been tested?

kaklik, 2020-04-27:
> 
> I also noticed that my Rascal doesn't have a smoke trail. Could this be something related to the model I have installed in my system?

There could be several possible differences.  I wrote some basic explanation in the new documentation: https://github.com/ThunderFly-aerospace/PX4Devguide/blob/flightgear/en/simulation/flightgear_vehicles.md

Jaeyoung-Lim, 2020-05-22:
@kaklik I still see this problem time to time, why was this marked as invalid?

kaklik, 2020-05-27:
I am unable to reproduce this behavivour using the up to date code.

slimonslimon, 2021-02-15:
there are lots of updates in PX4 and this bridge, is this problem still reproducable? Last update of this bridge stabilized gyroscope. So It is harder (or may not possible) to "confuse QGC horizont and compass" i.e. confuse PX4 position estimator.

Jaeyoung-Lim, 2021-02-15:
@slimonslimon Yes, this problem comes from the fact that the flightgear bridge is not on lockstep. Therefore if you have a laptop that is not powerful enough, you will reproduce this problem.