ThunderFly-aerospace/PX4-FlightGear-Bridge #11: Improve noise injection model

kaklik, 2020-04-28:
The current implementation has a simple Gaussian noise model injected to the sensor inputs. 

This simplified model could possibly bring problems with unrealistic sensor readout.  This may result in unpredictable behavior, which could be the primary problem of other issues: 

- #10 - flyaway, which is probably caused by multiple bugs occurring at the same time. The PX4 itself should be resistant to this. 

The solution of this issue is the implementation of similar noise model, which is used in Gazebo.

Jaeyoung-Lim, 2020-04-29:
@kaklik Would be great if this could be easily configurable. What are your plans on moving forward?

kaklik, 2020-04-29:
We do not plan to solve this issue in the near future.  Our development is not limited by the current implementation.