ThunderFly-aerospace/PX4-FlightGear-Bridge #15: Remove ign::math dependency

slimonslimon, 2020-05-01:
By selecting other Vector/quterernion library - Idealy some "header only", or code copy type.

Or write the rotation directly into code.

Jaeyoung-Lim, 2020-12-08:
@slimonslimon Any updates on this?

slimonslimon, 2020-12-08:
@Jaeyoung-Lim, Oh I didnt know about this bug i with ubuntu 20.04. I will look at this when I return from work.

Jaeyoung-Lim, 2021-01-04:
@slimonslimon Do you have any updates regarding this? I can see that you have updated the documentation, but IMHO it is not correct to use `ignition-math` as a dependency, since this repository is not using `ignition-transport`

slimonslimon, 2021-01-04:
@Jaeyoung-Lim Im sorry but I forgot about this, lot of other work and other problems.

What do you thing about copying few templates (vector, matrices, quaternion, ..) from ignition-math into this repository? (It is shortest way to get it work without external dependency and IMHO these basic template classes are programed correctly)

Jaeyoung-Lim, 2021-01-04:
@slimonslimon What features are you interested from `ignition-math`? I am a little surprised that you are interested in copying over the templates, since not much of the features of the math libraries are used in this repo - We can even just use eigen quaternions

slimonslimon, 2021-01-04:
"What features are you interested from ignition-math?"
 The reason for using Ignition-math was simple - before writing this, I was working with gazebo plugins, so i know API of ignition-math, so i didnt search for other library.

Thank you for pointing me to  Eigen library. If dependency on Eigen is better then ign-math I will look at how much is compatible (I hope, it will be simple to switch to this)

kaklik, 2021-01-06:
New Eigen3 library tested by: 

    make px4_sitl_nolockstep flightgear_rascal

woks as expected, closing.

closed 2021-01-06 by kaklik