ThunderFly-aerospace/PX4-FlightGear-Bridge #17: Ubuntu bionic flightgear SITL failure

Jaeyoung-Lim, 2020-05-02:
On Ubuntu 18.04, on a fresh system, flight gear SITL fails with the following error: 

```
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2020-05-02/11_18_50.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh>     0.20 [INFO]:OSG        Error: Not able to create requested visual.
    0.21 [ALRT]:view       getDefaultWindow: failed to create GraphicsContext
    0.21 [INFO]:OSG        Error: Not able to create requested visual.
    0.21 [ALRT]:view       getDefaultWindow: failed to create GraphicsContext
    0.21 [WARN]:view       CameraGroup::buildGUICamera: failed to build a window
    0.21 [WARN]:general    Enabling ATI/AMD viewport hack
    0.21 [INFO]:general    Configuration State
    0.21 [INFO]:general    ============= =====
    0.21 [INFO]:general    aircraft-dir = "/home/jaeyoung/src/Firmware/Tools/flightgear_bridge/models/FlightGear-Rascal"
    0.21 [INFO]:general    fghome-dir = "Path "/home/jaeyoung/.fgfs""
    0.21 [INFO]:general    download-dir = "/home/jaeyoung/.fgfs"
    0.21 [INFO]:general    terrasync-dir = "/home/jaeyoung/.fgfs/TerraSync"
    0.21 [INFO]:general    aircraft-search-paths = 
	/home/jaeyoung/src/Firmware/Tools/flightgear_bridge/models
    0.21 [INFO]:general    scenery-search-paths = 
	/home/jaeyoung/.fgfs/TerraSync
PX4 Communicator: PX4 Connected.
Bridge: Signal SIGINT recieve
```

Jaeyoung-Lim, 2020-05-02:
Switching the graphics drivers from `xserver-xorg-video-nouveau` to `nvidia-driver-435` fixed the problem

closed 2020-05-02 by Jaeyoung-Lim

bys1123, 2020-05-14:
Do you have any idea why run flightgear standalone will have no issue, why sitl side related graphics driver?

Jaeyoung-Lim, 2020-05-15:
@bys1123 Yes, because flightgear uses hardware acceleration to render the simulation