ThunderFly-aerospace/PX4-FlightGear-Bridge #18: Rascal looks differently

Jaeyoung-Lim, 2020-05-02:
I have removed the previously installed rascal model and tried running the fg nightly build.

However, the rascal model looks a bit different compared to the previously installed model. 
[image]

kaklik, 2020-05-02:
This can happen in the situation when some parts of the model are not found.  Yellow-blue airplane is a common simplification of FG model in situations where something gets wrong.   The command-line messages probably can help you debug this issue.

Jaeyoung-Lim, 2020-05-02:
@kaklik Oh, right. Thanks for clarifying.

The log is as below
```
I'm Mavlink to FlightGear Bridge
Targed Bridge Freq: 200, send data every step: 1
4
  5   -1
  7   -1
  2   1
  4   1
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [commander] Mission #3 loaded, 11 WPs, curr: 0
    0.04 [INFO]:gui        UI languages: ("en-US")
    0.04 [INFO]:gui        Loaded translations for locale "en_US"
    0.04 [INFO]:general    FlightGear:  Version 2019.2.0
    0.04 [INFO]:general    FlightGear:  Build Type Nightly
    0.04 [INFO]:general    Built with GNU C++ version 7.5
    0.04 [INFO]:general    Jenkins number/ID 0:none
    0.04 [INFO]:general    platform default fg_root = Path "/usr/share/games/flightgear"
INFO  [init] Mixer: etc/mixers-sitl/plane_sitl.main.mix on /dev/pwm_output0
    0.05 [INFO]:input      Reading user settings from Path "/home/jaeyoung/.fgfs/autosave_2019_2.xml"
    0.05 [INFO]:general    init package root at:Path "/home/jaeyoung/.fgfs/Aircraft"
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [airspeed_selector] No airspeed sensor detected. Switch to non-airspeed mode.
    0.06 [INFO]:input      aircraft = Rascal110-YASim
    0.06 [INFO]:general    Loading aircraft -set file from:Path "/home/jaeyoung/src/Firmware/Tools/flightgear_bridge/models/FlightGear-Rascal/Rascal110-YASim-set.xml"
    0.06 [ALRT]:aircraft   Aircraft does not specify a minimum FG version: please add one at /sim/minimum-fg-version
    0.06 [INFO]:general    Found language resource for: en_US
    0.06 [INFO]:general    Reading localized strings for 'en' from Path "/usr/share/games/flightgear/Translations/default/sys.xml"
    0.06 [INFO]:general    Reading localized strings for 'en' from Path "/usr/share/games/flightgear/Translations/default/atc.xml"
    0.06 [INFO]:general    Reading localized strings for 'en' from Path "/usr/share/games/flightgear/Translations/default/tips.xml"
    0.06 [INFO]:general    	option:aircraft = Rascal110-YASim
    0.06 [INFO]:general    	option:fg-aircraft = ./models
    0.06 [INFO]:general    	option:enable-terrasync = 
    0.06 [INFO]:general    	option:timeofday = noon
    0.06 [INFO]:general    	option:disable-sound = 
    0.06 [INFO]:general    Channel string = 15400
    0.06 [INFO]:general    	option:telnet = 15400
    0.06 [INFO]:general    Channel string = socket,out,100,127.0.0.1,15200,udp,FGtoPX4
    0.06 [INFO]:general    	option:generic = socket,out,100,127.0.0.1,15200,udp,FGtoPX4
    0.06 [INFO]:general    Channel string = socket,in,100,,15300,udp,PX4toFG
    0.06 [INFO]:general    	option:generic = socket,in,100,,15300,udp,PX4toFG
    0.06 [INFO]:general    	option:model-hz = 120
    0.06 [INFO]:general    	option:disable-random-objects = 
    0.06 [INFO]:general    	option:prop = /sim/rendering/texture-compression=off
    0.06 [INFO]:general    	option:prop = /sim/rendering/quality-level=0
    0.06 [INFO]:general    	option:prop = /sim/rendering/shaders/quality-level=0
    0.06 [INFO]:general    	option:disable-ai-traffic = 
    0.06 [INFO]:general    	option:prop = /sim/ai/enabled=0
    0.06 [INFO]:general    	option:prop = /sim/rendering/random-vegetation=0
    0.06 [INFO]:general    	option:prop = /sim/rendering/random-buildings=0
    0.06 [INFO]:general    	option:disable-specular-highlight = 
    0.06 [INFO]:general    	option:disable-ai-models = 
    0.06 [INFO]:general    	option:disable-clouds = 
    0.06 [INFO]:general    	option:disable-clouds3d = 
    0.06 [INFO]:general    	option:fog-fastest = 
    0.06 [INFO]:general    	option:visibility = 2000
    0.06 [INFO]:general    	option:disable-distance-attenuation = 
    0.06 [INFO]:general    	option:disable-real-weather-fetch = 
    0.06 [INFO]:general    	option:prop = /sim/rendering/particles=0
    0.06 [INFO]:general    	option:prop = /sim/rendering/multi-sample-buffers=1
    0.06 [INFO]:general    	option:prop = /sim/rendering/multi-samples=2
    0.06 [INFO]:general    	option:prop = /sim/rendering/draw-mask/clouds=false
    0.06 [INFO]:general    	option:prop = /sim/rendering/draw-mask/aircraft=true
    0.06 [INFO]:general    	option:prop = /sim/rendering/draw-mask/models=true
    0.06 [INFO]:general    	option:prop = /sim/rendering/draw-mask/terrain=true
[... 82 lines cut ...]
YASim warning: possible convergence problem.
FGBridge: Low FlightGear Update Freq: 5.71429
    8.03 [INFO]:OSG        
    8.03 [WARN]:OSG        PNG lib warning : iCCP: known incorrect sRGB profile
    8.04 [WARN]:OSG        PNG lib warning : iCCP: known incorrect sRGB profile
WARN  [airspeed_selector] Airspeed: switched from sensor -1 to 1
INFO  [ecl/EKF] 9299022871: GPS checks passed (WGS-84 origin set)
    8.47 [WARN]:OSG        PNG lib warning : iCCP: known incorrect sRGB profile
    8.49 [WARN]:OSG        PNG lib warning : iCCP: known incorrect sRGB profile
    8.49 [WARN]:OSG        PNG lib warning : iCCP: known incorrect sRGB profile
    8.49 [WARN]:OSG        PNG lib warning : iCCP: known incorrect sRGB profile
    8.49 [WARN]:OSG        PNG lib warning : iCCP: known incorrect sRGB profile
    8.52 [WARN]:OSG        PNG lib warning : iCCP: known incorrect sRGB profile
    8.52 [WARN]:OSG        PNG lib warning : iCCP: known incorrect sRGB profile
[... cut ...]
    8.52 [WARN]:OSG        PNG lib warning : iCCP: known incorrect sRGB profile
    8.53 [WARN]:OSG        PNG lib warning : iCCP: known incorrect sRGB profile
INFO  [ecl/EKF] 9299788114: EKF aligned, (baro height, IMU buf: 18, OBS buf: 14)
INFO  [ecl/EKF] 9299788114: reset position to last known position
INFO  [ecl/EKF] 9299788114: reset velocity to zero
INFO  [ecl/EKF] 9303612347: reset position to GPS
INFO  [ecl/EKF] 9303612347: reset velocity to GPS
INFO  [ecl/EKF] 9303612347: commencing GPS fusion

```

kaklik, 2020-05-02:
It seems you have defined the model path as a relative path. 
`option:fg-aircraft = ./models`
FG does not like relative paths.

Jaeyoung-Lim, 2020-05-02:
I didn't change anything out of px4 master. Seems like that is coming from https://github.com/ThunderFly-aerospace/PX4-FlightGear-Bridge/blob/master/FG_run.py#L42

kaklik, 2020-05-02:
> I didn't change anything out of px4 master. Seems like that is coming from https://github.com/ThunderFly-aerospace/PX4-FlightGear-Bridge/blob/master/FG_run.py#L42

Ok, it seems to be a bug in the FG_run.py.   Apparently you can work around it by setting the FG_MODELS_DIR variable to the required absolute path to the directory of models. 

You should be inspired by [this script](https://github.com/ThunderFly-aerospace/PX4-FlightGear-Bridge/blob/master/scripts/vehicle1.sh#L6).

saengphet, 2020-05-03:
Hi, I got the same problem "failed to find aircraft model".

Jaeyoung-Lim, 2020-05-03:
@kaklik Updating the submodule to the master of this repo solved the problem. This needs a backport as https://github.com/PX4/PX4-FlightGear-Bridge/pull/3 and a update on the firmware side

kaklik, 2020-05-03:
> 
> seems like [eb89f37](https://github.com/ThunderFly-aerospace/PX4-FlightGear-Bridge/commit/eb89f37c7fe69efc286d8bce8bedb3f1541f9db8) fixes it?

That commit could solve the issue in the case where you had the Rascal model directory named by old name Rascal110. 

In another case, we should check that FG_run.py  is completely avoiding to use a relative path in `fgfs` startup parameters.

Jaeyoung-Lim, 2020-05-03:
@kaklik This is not a special setup for me. I am trying to debug the broken setup in master of PX4/firmware.

It seems like master has a model name as `FlightGear-Rascal` (https://github.com/PX4/PX4-FlightGear-Bridge/tree/13dde0b4a42be19e9e9dbbb1c2527fab634112af/models) and this has been changed Rascal110 then to Rascal

saengphet, 2020-05-03:
> @kaklik @saengphet Updating the submodule to the master of this repo solved the problem. This needs a backport as [PX4#3](https://github.com/PX4/PX4-FlightGear-Bridge/pull/3) and a update on the firmware side
> 
> seems like [eb89f37](https://github.com/ThunderFly-aerospace/PX4-FlightGear-Bridge/commit/eb89f37c7fe69efc286d8bce8bedb3f1541f9db8) fixes it?

Solved, thank you !!

closed 2020-05-03 by Jaeyoung-Lim