ThunderFly-aerospace/PX4-FlightGear-Bridge #21: Can’t run simulation with FightGear in Mac

kevinuav, 2020-06-12:
Hi,everybody.
I am tiring to make the simulation of FlighrGear with my mac. Although I do have installed FlightGear(2020.1.2), and I have found the file"fgfs" in the installed package(FLightGear->package contents->MacOS->fgfs) So the “fgfs” is not located in the same directory as in the ubuntu. And I can’t run it at the command line. I can just run it with double clicking the “fgfs” icon…It still encounters errors. Dose PX4 not support FlightGear for mac yet? Or is it just my setting problem?
here is the error message:

kaklik, 2020-06-12:
FlightGear-bridge is completely untested on mac. We are not using Apple in our develompent pipeline.

slimonslimon, 2020-06-17:
FG_BINARY="-- /Applications/FlightGear.app/Contents/MacOS/fgfs"

Why is there "--" at begin of FG_BINARY parameter? Is it somemething MacOS specific?

kaklik, 2022-01-06:
Is it still a relevant issue? I want to close it, because the root of the trouble seems to not be related to using Mac.

closed 2022-01-10 by kaklik