ThunderFly-aerospace/PX4-FlightGear-Bridge #22: Zigzags in flight path in QGroundControl (Plane teleports)

lum-supakorn, 2020-06-18:
**Steps to reproduce:** 
1. `make px4_sitl_nolockstep flightgear_rascal-electric` or `make px4_sitl_nolockstep flightgear_rascal`
2. Takeoff from QGroundControl and watch the plane loiter

From time to time the plane in QGroundControl will teleport and the flight path will be drawn as shown in the figure below:

[image]

Note that the plane in FlightGear does not teleport, so I assume that the issue comes from PX4 FlightGear bridge.

slimonslimon, 2020-06-18:
Two Quietions: 
1) What FPS do you have in FlightGear?
2) Are you able to publish Log from flight ( at least text log from console?)

lum-supakorn, 2020-06-19:
@slimonslimon The frame rate swings from 50 to 100 FPS.
Somehow I cannot collect the flight log from QGroundControl, but here is the text log from the console (not the same flight as that in the question but there was still teleportation):
```
[0/4] Performing build step for 'flightgear_bridge'
ninja: no work to do.
[3/4] cd /home/lum/Code/PX4/Firmware/b...PX4/Firmware/build/px4_sitl_nolockstep
SITL ARGS
sitl_bin: /home/lum/Code/PX4/Firmware/build/px4_sitl_nolockstep/bin/px4
debugger: none
program: flightgear
model: rascal-electric
world: none
src_path: /home/lum/Code/PX4/Firmware
build_path: /home/lum/Code/PX4/Firmware/build/px4_sitl_nolockstep
FG setup
    0.08 [INFO]:gui        UI languages: ("en-US")
    0.08 [INFO]:gui        Loaded translations for locale "en_US"
    0.08 [INFO]:general    FlightGear:  Version 2020.1.1
    0.08 [INFO]:general    FlightGear:  Build Type Dev
    0.08 [INFO]:general    Built with GNU C++ version 9.3
    0.08 [INFO]:general    Jenkins number/ID 0:none
    0.08 [INFO]:general    platform default fg_root = Path "/usr/share/games/flightgear"
    0.10 [INFO]:input      Reading user settings from Path "/home/lum/.fgfs/autosave_2020_1.xml"
    0.10 [INFO]:general    init package root at:Path "/home/lum/.fgfs/Aircraft"
    0.10 [INFO]:input      No user specified aircraft, using default
    0.10 [INFO]:general    Loading aircraft -set file from:Path "/usr/share/games/flightgear/Aircraft/c172p/c172p-set.xml"
    0.13 [INFO]:general    Found language resource for: en_US
    0.13 [INFO]:general    Reading localized strings for 'en' from Path "/usr/share/games/flightgear/Translations/default/sys.xml"
    0.13 [INFO]:general    Reading localized strings for 'en' from Path "/usr/share/games/flightgear/Translations/default/atc.xml"
    0.13 [INFO]:general    Reading localized strings for 'en' from Path "/usr/share/games/flightgear/Translations/default/tips.xml"
    0.13 [INFO]:general    	option:version = 
    0.13 [INFO]:general    Using default download dir: Path "/home/lum/.fgfs"
    0.13 [INFO]:general    Using default texture cache directory: Path "/home/lum/.fgfs/TextureCache"
    0.13 [INFO]:general    Using TerraSync dir: Path "/home/lum/.fgfs/TerraSync"
    0.13 [INFO]:io         Saving user settings to Path "/home/lum/.fgfs/autosave_2020_1.xml"
    0.14 [WARN]:general    SGSubsystemMgr::remove: not found: scenery
None
Rascal110-Electric-YASim
https://github.com/ThunderFly-aerospace/FlightGear-Rascal/archive/master.zip
5 /controls/flight/aileron -1
7 /controls/flight/elevator -1
2 /controls/flight/rudder 1
4 /controls/engines/engine/throttle 1
fgfs  --aircraft=Rascal110-Electric-YASim --fg-aircraft=./models --enable-terrasync --timeofday=noon --disable-sound --telnet=15400 --generic=socket,out,100,127.0.0.1,15200,udp,FGtoPX4 --generic=socket,in,100,,15300,udp,PX4toFG --model-hz=120 --disable-random-objects --prop:/sim/rendering/texture-compression=off --prop:/sim/rendering/quality-level=0 --prop:/sim/rendering/shaders/quality-level=0 --disable-ai-traffic --prop:/sim/ai/enabled=0 --prop:/sim/rendering/random-vegetation=0 --prop:/sim/rendering/random-buildings=0 --disable-specular-highlight --disable-ai-models --disable-clouds --disable-clouds3d --fog-fastest --visibility=2000 --disable-distance-attenuation --disable-real-weather-fetch --prop:/sim/rendering/particles=0 --prop:/sim/rendering/multi-sample-buffers=1 --prop:/sim/rendering/multi-samples=2 --prop:/sim/rendering/draw-mask/clouds=false --prop:/sim/rendering/draw-mask/aircraft=true --prop:/sim/rendering/draw-mask/models=true --prop:/sim/rendering/draw-mask/terrain=true --disable-random-vegetation --disable-random-buildings --disable-rembrandt --disable-horizon-effect & echo $! > /tmp/px4fgfspid_0
SITL COMMAND: "/home/lum/Code/PX4/Firmware/build/px4_sitl_nolockstep/bin/px4" "/home/lum/Code/PX4/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/lum/Code/PX4/Firmware"/test_data
INFO  [px4] Creating symlink /home/lum/Code/PX4/Firmware/ROMFS/px4fmu_common -> /home/lum/Code/PX4/Firmware/build/px4_sitl_nolockstep/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [param] selected parameter default file eeprom/parameters_1034
[param] Loaded: eeprom/parameters_1034
I'm Mavlink to FlightGear Bridge
Targed Bridge Freq: 200, send data every step: 1
4
  5   -1
  7   -1
  2   1
  4   1
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [commander] LED: open /dev/led0 failed (22)
[... 166 lines cut ...]
FGBridge: Low FlightGear Update Freq: 17.1429
FGBridge: Low FlightGear Update Freq: 20
INFO  [ecl/EKF] 1349342267: reset velocity to GPS
INFO  [ecl/EKF] 1349342267: reset position to GPS
INFO  [ecl/EKF] 1349342267: Emergency yaw reset - mag use stopped
FGBridge: Low FlightGear Update Freq: 17.1429
INFO  [ecl/EKF] 1406762730: reset velocity to GPS
INFO  [ecl/EKF] 1406762730: reset position to GPS
INFO  [ecl/EKF] 1406762730: Emergency yaw reset - mag use stopped
FGBridge: Low FlightGear Update Freq: 17.1429
INFO  [navigator] RTL LAND activated
INFO  [navigator] RTL: landing at home position.
[... cut ...]
INFO  [navigator] RTL: return at 145 m (100 m above destination)
INFO  [navigator] RTL: descend to 136 m (91 m above destination)
INFO  [navigator] RTL: completed, loitering
INFO  [fw_pos_control_l1] Landing, on slope
INFO  [ecl/EKF] 1430538618: reset velocity to GPS
INFO  [ecl/EKF] 1430538618: reset position to GPS
WARN  [ecl/EKF] 1430538618: GPS fusion timeout - reset to GPS
INFO  [commander] Landing detected
INFO  [commander] DISARMED by Auto disarm initiated
INFO  [logger] closed logfile, bytes written: 13203301
```

kaklik, 2020-06-19:
You have multiple occurrences of following lines in the console output: 

     FGBridge: Low FlightGear Update Freq: 17.1429

It seems there is trouble with the FlightGear frame rate..  Are you able to check the frame rate of FG itself by [enabling the FPS display](https://forum.flightgear.org/viewtopic.php?f=37&t=11163)?

lum-supakorn, 2020-06-19:
@kaklik I did obtain the frame rate of FG through FPS display you mentioned. Do you know what caused the ` FGBridge: Low FlightGear Update Freq` prompt?

slimonslimon, 2020-06-24:
The bridge internally counts how often comes messages from the FG. It can be caused by "some interference" of Frequency of internal pooling loop of bridge: https://github.com/ThunderFly-aerospace/PX4-FlightGear-Bridge/blob/master/src/flightgear_bridge.cpp#L145 and frequency of FlightGear main loop (somehow connected with FG FPS - FG is single thread).

To proof this hypothesis, you can change frequency of bridge here: https://github.com/ThunderFly-aerospace/PX4-FlightGear-Bridge/blob/master/src/flightgear_bridge.cpp#L98

slimonslimon, 2020-06-24:
But problems can be at more places and they are hard to reproduce or debug. 
Try reduce mentioned delay closer to zero - it will increase bridge frequency. For example to 100 (theroreticly 10KHz - but it will be slowed down by poling of sockets) and switch off bad data upsampling - set https://github.com/ThunderFly-aerospace/PX4-FlightGear-Bridge/blob/master/src/flightgear_bridge.cpp#L101 to false.

Then the bridge should pool for messages from FG very often and only resend message to PX4. (I hope that it will be working) 

The last think what you can set up better is requested frequency of messages from FlightGear - now it is set to 100 Hz. But FG is single thread aplication, so real Frequency of messages will be somehow connected with FPS - it will not be large than FPS. So if you request 1000 messages per second, It will be sending message every frame.
You can set-up this a here: https://github.com/ThunderFly-aerospace/PX4-FlightGear-Bridge/blob/master/FG_run.py#L104 
rewrite the number 100 with better number. And do the same substitution on next line.

Thank you for your time that you spend with testing these things.

lum-supakorn, 2020-06-29:
@slimonslimon Thank you for your input and apologize for late reply. I will look into this further as soon as I can.

roman-dvorak, 2020-07-02:
I'm not able to reproduce this zigzaging

[image]

Jaeyoung-Lim, 2020-07-10:
I think this points to the fact that we need lockstep for flightgear

lum-supakorn, 2020-07-11:
@Jaeyoung-Lim Agreed! Lockstep also makes debugging models much easier.

slimonslimon, 2020-07-11:
@Jaeyoung-Lim @lum-supakorn  - It could be wery cool - but there are 2 main problems:
  - First one - I dont know how to do it easy ( I think It will need some improvements on FG side)
  - Second one - day have only 24h....

Jaeyoung-Lim, 2020-07-12:
@slimonslimon Right, but just pointing out that this part of SITL will never be reliable without it.

If we are not able to support it, the least we should do is provide strict performance requirements on hardware to run the simulation.

lum-supakorn, 2020-07-14:
@Jaeyoung-Lim Does JSBSim not support something similar to lockstep?

Jaeyoung-Lim, 2020-07-14:
JSBSim does, but the flight gear interfaced used in this repo does not

kaklik, 2020-07-14:
@Jaeyoung-Lim  @lum-supakorn  I suggest continuing at the discussion about the lock-step option in [different issue: #8](https://github.com/ThunderFly-aerospace/PX4-FlightGear-Bridge/issues/8) Where I also commented on the actual possibilities.

slimonslimon, 2020-07-14:
@lum-supakorn are you able to test modifications which was mentitioned above? (Because exactly I have not hardware which has higer FPS than about 50)

slimonslimon, 2021-02-15:
well, after long time I have improved gyroscope readouts which may solve this issue. 
https://github.com/ThunderFly-aerospace/PX4-FlightGear-Bridge/commit/d7470250d9a45439957b0e5c0e123900e44d1d29