ThunderFly-aerospace/PX4-FlightGear-Bridge #25: Aircraft paths are not compatible with jsbsim

Jaeyoung-Lim, 2020-08-29:
I can see that the aircraft configs under `models` directory in this repo is not compatible with the path that JSBSim is assuming. 

See: https://github.com/JSBSim-Team/jsbsim/blob/ecb79520a50d0f04ca095801f39c3bdb3ea36a16/src/FGFDMExec.h#L280
```
  /** Loads an aircraft model.  The paths to the aircraft and engine
      config file directories must be set prior to calling this.  See
      below.
      @param model the name of the aircraft model itself. This file will
      be looked for in the directory specified in the AircraftPath variable,
      and in turn under the directory with the same name as the model. For
      instance: "aircraft/x15/x15.xml"
      @param addModelToPath set to true to add the model name to the
      AircraftPath, defaults to true
      @return true if successful*/
  bool LoadModel(const std::string& model, bool addModelToPath = true);
```

As far as I understand this is handled in the json file when launching with the python script, but seems like this doesn't work when using the cpp API of jsbsim.

Any ideas on how to get this working? @kaklik @slimonslimon  

@RomanBapst FYI

kaklik, 2020-08-31:
Name the aircraft XML file exactly the same as the aircraft directory is highly impractical.  Because many aircraft models have multiple variants (which is common to FlightGear to select it by XML file name).

Therefore the best solution would be a workaround in JSBSim API.

Jaeyoung-Lim, 2020-10-15:
This was fixed in https://github.com/Auterion/px4-jsbsim-bridge/pull/10 with the jsbsim integration separately.

closed 2020-10-15 by Jaeyoung-Lim