ThunderFly-aerospace/PX4-FlightGear-Bridge #27: Primary EKF changed X (unhealthy) -> X

kaklik, 2021-01-06:
A suspicious warning message emerged in the terminal after running the simulation. 

[image]

The effect on simulator performance is unknown to me. It seems to works correctly.

roman-dvorak, 2021-01-14:
It is possible that this is causing this problem - bad estimation of altitude. 
[image]

Baro and GPS data from FG looks fine. But estimation is bad. Interesting is, that jitter offset is always same (200 m).

roman-dvorak, 2021-01-15:
The problem from the last comment looks solved by disabling multiple EKF (https://github.com/PX4/PX4-Autopilot/pull/14650). 

New parameter configuration is here:
[image]

So far, I am not able to repeat it with another model.
It seems not to be related to FG FPS. 

@kaklik, can you try to reproduce this behaving with parameters:
`ekf2_multi_imu` : 3
`sens_imu_mode`: 0
In FG sitl with TF-G1 master. [
So far, I am not able to repeat it with another model (FW) ](url)

@dagar Please, Is possible that one of the multiple (Multi-EKF) estimators estimate a different altitude. And constant switching sometimes comes across a "problematic" estimator?

dagar, 2021-01-15:
Can you share the full log? https://logs.px4.io/

roman-dvorak, 2021-01-15:
Yes, log file is here: 
https://logs.px4.io/plot_app?log=b3c6ce24-aad7-4040-8c7e-cbcbf98b1fb1

dagar, 2021-01-16:
2 issues (so far)

1. I believe the rapid switching should be resolved with https://github.com/PX4/PX4-Autopilot/pull/16518. The logging for this is also a lot better in current master (https://github.com/PX4/PX4-Autopilot/pull/16529). In 
[image]

2. the difference in altitude estimates per estimator
      - only 1 out of 4 logged ekf2 instances have a different altitude estimate
[image]

I believe the offset in z estimates can be explained by the baro altitude ramping combined with slightly different timing of height resets.

[image]

Note, this isn't actually a problem as long as all consumers of estimator data respect the resets, althogh I'd still like to minimize these differences as much as possible. https://github.com/PX4/PX4-Autopilot/blob/1c0ed74d6a91d39cffbddf8ff5e6a8e1c329ebb1/msg/vehicle_local_position.msg#L20-L21

slimonslimon, 2021-01-17:
`baro hgt timeout - reset to baro` - what frequency of messages is expected from baro? (or where can i found it in code?)
(or I missunderstand the `baro hgt timeout` message?)

FG sends its messages into bridge maximaly at FPS rate, this bridge can do some upsampling by resending same message with random noise - (normal distribution, but t dont know what variance shoud be set - i set it somehow....)

dagar, 2021-01-24:
> `baro hgt timeout - reset to baro` - what frequency of messages is expected from baro? (or where can i found it in code?)
> (or I missunderstand the `baro hgt timeout` message?)
>
> FG sends its messages [...]

Low rate (eg 10 Hz) is fine as long as the jitter isn't horrible. The message is perhaps a bit misleading because it's not necessarily a timeout (gap in data), it's more of a general problem with that height source (like high baro innovations).

dagar, 2021-01-24:
This bug in the ekf2selector also may have been a factor. https://github.com/PX4/PX4-Autopilot/pull/16622
Even though you had estimators that ended up with height offsets, switching between them should still have been handled smoothly by the controllers.

slimonslimon, 2021-01-28:
@dagar thank you for your comments! 

It seems that baro ramping is caused by FG inicialization (FG probabyl models The Big bang...) I will put an 5sec delay, before start sending messages into PX4.

But "Primary EKF changed X (unhealthy) -> X" is still unresolved. Maybe increase of sending messages to PX4 produce less messages but, im not sure.

kaklik, 2021-01-31:
> 
> @kaklik, can you try to reproduce this bug with the use of the following parameters?
> 
> ```
> ekf2_multi_imu : 3
> sens_imu_mode`: 0
> ```

Yes, the issue happens in the case where the multiple IMUs are activated with SITL simulation.  Disabling multiple IMUs by setting:

     ekf2_multi_imu : 1
     sens_imu_mode: 1
 
solves the problem.  But in the case of updated PX4Firmware/master, a new similar problem emerges:

    ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 22378 -> 22381
    ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 22383 -> 22386
    ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 22386 -> 22388
    ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 22389 -> 22393
    ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 22395 -> 22398
    ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 22400 -> 22402
    ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 22404 -> 22406
    ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 22408 -> 22410
    ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 22410 -> 22412

The PX4 generate thousands of such messages.  The message is [generated here](https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/ekf2/EKF2.cpp#L1425). But I do not understand the meaning of this message.

dagar, 2021-01-31:
> The PX4 generate thousands of such messages. The message is generated here. But I do not understand the meaning of this message.

It means `ekf2` is isn't keeping up with `vehicle_magnetometer` publications for some reason. This is a new debugging message only recently added to master to find potential problems and it needs to be investigated.

dagar, 2021-01-31:
> It means `ekf2` is isn't keeping up with `vehicle_magnetometer` publications for some reason. This is a new debugging message only recently added to master to find potential problems and it needs to [...]

Should be fixed by https://github.com/PX4/PX4-Autopilot/pull/16698.

slimonslimon, 2021-02-01:
Thank you @dagar ! It seem to work!

kaklik, 2021-02-01:
The FlightGear-PX4 working setup has been reached again. Required parameters setup: 

     ekf2_multi_imu : 1
     sens_imu_mode: 1
 
Altitude ramp up solved by delay at beginning of the simulation. It is tested with current PX4upstream/master firmware. 

Thanks to all! :)

closed 2021-02-01 by kaklik

BrenoPimentel, 2021-08-17:
I got the same error trying to write PX4 commands on prompt, however, i just restart my PC and launch the software again and well done. It works normally :)

lygoon, 2022-11-30:
I also encountered this error（WARN  [ekf2] primary EKF changed 0 (unhealthy) -> 2）, but I don't know where to correct it. Which file can I modify? thank you!

kaklik, 2022-11-30:
> I also encountered this error（WARN [ekf2] primary EKF changed 0 (unhealthy) -> 2）, but I don't know where to correct it. Which file can I modify? thank you!

Please check, that you are running the up-to-date FlightGear-Bridge code.  You could also disable this message by disabling ekf2_multi_imu in PX4 parameters.