ThunderFly-aerospace/PX4-FlightGear-Bridge #31: Does PX4 deprecate PX4-FlightGear-Bridge simulation?

kevinuav, 2023-10-16:
Hi, I am using the simulation with the latest firmware. However. I encounters errors while it simulates with flight gear. Does px4 deprecate px4-flightgear-bridge? It just responds like this:

PX4-Autopilot % make px4_sitl_nolockstep flightgear_rascal
ninja: error: unknown target 'flightgear_rascal'
even the command:

PX4-Autopilot % make px4_sitl list_vmd_make_targets             
ninja: error: unknown target 'list_vmd_make_targets'
make: *** [px4_sitl] Error 1

kaklik, 2023-10-16:
This unmerged pull-request should resolve that: https://github.com/PX4/PX4-Autopilot/pull/21853

slimonslimon, 2023-10-16:
Hi, do you have Flightgear installed? Does fgfs command work?

In build system of PX4 are some conditions which disable FG support when command fgfs binnary is not available. (simlink is not enought, i dont know why)

(Will test current state later)

kevinuav, 2023-10-17:
I have installed FlightGear already. And it can be launch with command"% /Applications/FlightGear.app/Contents/MacOS/fgfs"

kaklik, 2023-10-17:
> "% /Applications/FlightGear.app/Contents/MacOS/fgfs"

Ok, that is the cause of the issue. The PX4 make system expects that the simple "fgfs" command could run FlightGear. Therefore you should use something like PATH variable or a hardlink. (symlink is not enough, as @slimonslimon  posted above.)