ThunderFly-aerospace/PX4-FlightGear-Bridge #32: The error occurs when using FlightGear for software-in-the-loop simulation in v1.14 and later

747640013, 2024-01-12:
I use FlightGear for fixed-wing software-in-the-loop simulation. However, after running the command "make px4_sitl_nolockstep flightgear_rascal-electric," I encountered the following error:

```bash
ERROR [simulator_mavlink] Number of simulated accelerometers 80 out of range. Max: 3.
```

This issue arises in PX4 Firmware version 1.14, while it doesn't occur in version 1.13.3.  I am looking to resolve this problem, but I am having difficulty identifying the issue. Does anyone have any suggestions?

kaklik, 2024-01-12:
You should reduce the number of EKF filters running simultaneously in the PX4 parameters configuration. There may be an uninitialized variable.

747640013, 2024-01-13:
Thank you for your reminder. There should be uninitialized variables, as the number of accelerometer errors is inconsistent every time I start this simulation. I'll try to see if I can identify and resolve the issue.

747640013, 2024-01-13:
Thank you, it's just that the PX4 source code lacks initialization for sensor_msg.id

closed 2024-01-13 by 747640013